schreibfaul1/ESP32-audioI2S #1141: -Werror=misleading-indentation triggered when compiling using master branch

HAmamiya, 2025-08-19:
**Environment:**
-Windows
-VS code with [arduino-esp32 v3.3.0](https://github.com/espressif/arduino-esp32/releases/tag/3.3.0) & [ESP-IDF 5.5](https://github.com/espressif/esp-idf/tree/release/v5.5)

**Console output during compiling:**
```
H:/Project/components/ESP32-audioI2S/src/Audio.cpp: In member function 'int32_t Audio::newInBuffStart(int32_t)':
H:/Project/components/ESP32-audioI2S/src/Audio.cpp:6542:13: error: this 'if' clause 
does not guard... [-Werror=misleading-indentation]
 6542 |             if(m_codec == CODEC_M4A) m_resumeFilePos += m4a_correctResumeFilePos(); {if(m_resumeFilePos == -1) goto exit;}
      |             ^~
H:/Project/components/ESP32-audioI2S/src/Audio.cpp:6542:85: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
 6542 |             if(m_codec == CODEC_M4A) m_resumeFilePos += m4a_correctResumeFilePos(); {if(m_resumeFilePos == -1) goto exit;}
      |                                                                                     ^   
cc1plus.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```

**Recommended fix:**
Change `Audio.cpp:6542` to
```
            if(m_codec == CODEC_M4A) 
				m_resumeFilePos += m4a_correctResumeFilePos(); 
			if(m_resumeFilePos == -1) goto exit;
```

enjoyneering, 2025-08-19:
I'm not an expert, but in my projects I've sometimes noticed very strange compiler behavior if there are no "{}" brackets in the "if-else" statement.

`if (m_codec == CODEC_M4A) {m_resumeFilePos += m4a_correctResumeFilePos();}
if (m_resumeFilePos == -1) {goto exit;}`

schreibfaul1, 2025-08-20:
I'm no expert either, but I think that's what it means:
if(a) b; c;
Without brackets, we always execute c.

vortigont, 2025-08-20:
No, in this case `if(a) b; c;` it will execute `b` only `if (a)` evaluates **and then** it will _always_ execute `c;` as expected.
The compiler just notifies you that lexically it might _seem to a reader_ that `if(a) b; c;` would execute **both `b` and `c`** only if `a` evaluates to true, while it is not.
So for clear readability you should wire it as
```c++
if(a) b;
c;
```
or (better)
```c++
if(a)
  b;
c;
```

**OR** if you mean the other case, then use
```c++
if(a) { b; c; }
```

HAmamiya, 2025-08-20:
I would like to clarify that my suggested fix is not adopted by the owner.

My suggested fix is just the logic of the offended code, and in fact the owner wish to do something else. And therefore there is a fix.

enjoyneering, 2025-08-20:
> The problem with the first version is that if you go back and add a second statement to the if or else clauses without remembering to add the curly braces, your code will break in unexpected and amusing ways.

https://stackoverflow.com/questions/2125066/is-it-a-bad-practice-to-use-an-if-statement-without-curly-braces

closed 2025-09-20 by HAmamiya