schreibfaul1/ESP32-audioI2S #1143: Station stopped working after latest commits

cniedzi, 2025-08-20:
Hi Wolle (@schreibfaul1 ),

This station https://waw.ic.smcdn.pl/2380-1.aac (and others starting with this address) stopped playing after one of the latest commits. 
It still plays correctly in VLC.

Thanks!

hevet, 2025-08-21:
@HAmamiya For now, use this address https://ic1.smcdn.pl/2380-1.aac

HAmamiya, 2025-08-21:
I have further checked other few radio stations using icecast, e.g.
https://radio.wanderingsheep.net:8000/jazzcafe
http://streaming.intermediacolombia.com:18007/stream

And only https://waw.ic.smcdn.pl/2380-1.aac has this particular issue. What I guess is either the icecast powering this particular radio station is too old or there are some server side networking issues (e.g. server is behind proxy).

cniedzi, 2025-08-21:
The other issue I've just notice is that audio_info reports wrong stream parameters for [https://ic1.smcdn.pl/2380-1.aac](https://ic1.smcdn.pl/2380-1.aac):

```
audio_info: connect to: "ic1.smcdn.pl" on port 443 path "/2380-1.aac"
audio_info: SSL has been established in 259 ms
audio_info: icy-bitrate: 64000
audio_info: AACDecoder has been initialized
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 24000
audio_info: BitsPerSample: 16
audio_info: BitRate: 57000
audio_info: AAC HeaderFormat: ADTS
```

while in fact it is 32bit @ 48kHz...

closed 2025-08-24 by cniedzi