schreibfaul1/ESP32-audioI2S #1167: Connection issues for some stations required redirection

cniedzi, 2025-09-20:
Hi Wolle (@schreibfaul1 ),

Would you be so kind as to look into the following issue?

I’m trying to use the LovyanGFX graphics library together with your Audio library. Most stations work fine, but for some stations — the ones that require redirection to start streaming — I run into a problem::
```
Audio-info: audioI2S Version 3.4.2p
Audio-info: inputBufferSize: 655349 bytes
Audio-info: connect to: "getradio.reconv.pl" on port 443 path "/openfm?s=po-polsku-80"
[  3420][E][NetworkManager.cpp:136] hostByName(): DNS Failed for 'getradio.reconv.pl' with error '-54'
[  3440][E][ssl_client.cpp:120] start_ssl_client(): connect on fd 44, errno: 118, "Host is unreachable"
[  3450][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
Audio-LOGE: Audio.cpp:713: Request https://getradio.reconv.pl/openfm?s=po-polsku-80 failed!
```
When I use the standard Adafruit GFX library instead of LovyanGFX, everything works perfectly.

The errors above indicate network issues, and I have no idea how they could be related to the graphics library. I could understand memory issues, but not network problems...

**Important notice**: This issue applies to ESP32-S3 N16R8 only. When used with ESP32-P4 N16P32, there is no such a problem...

You can find a GitHub repository with a test project demonstrating this issue here: [https://github.com/cniedzi/Test_AudioI2S_LovyanGFX/tree/main](https://github.com/cniedzi/Test_AudioI2S_LovyanGFX/tree/main).

If you have some time, I would be very grateful for your insights.

cniedzi, 2025-09-21:
> That's just a guess. I think the heap is larger on the P4. SSL needs between 40 and 50 KB. You can move mbedtls to PSRAM in menuconfig. Maybe that will help.

I'm afraid it doesn't help because it has already been there :(
[image]
Changing it also doesn't help.

Currently, I'm not getting the network error message, but the following one:
```
Audio-info: inputBufferSize: 655349 bytes
Audio-info: connect to: "getradio.reconv.pl" on port 443 path "/openfm?s=po-polsku-80"
Audio-info: SSL has been established in 430 ms
Audio-lasthost: https://getradio.reconv.pl/openfm?s=po-polsku-80
Audio-info: chunked data transfer
Audio-info: redirect to new host "https://stream-cdn-1.open.fm/OFM20/ngrp:standard/playlist.m3u8?t=cbb91a479b547ab4884e1ea7c11899d319925706a865e3a1ddc00e5aef96a73cacf30f047c70506b7c7b74df725dec
                                                                                                                                                                                                ce7c15b976aeff1f2be8ad0a36f2782f8586c3786cb424328da1aff1d38808dcaacc6b1c295a26deda4d3f6901e97918663136727c67b815f1ff04ace3d0e87f7ff1f9fe669de2ebf072ac2cd34cb6966d86501d4321488c3767e7bc5a60cae5296fcbb186a55cb1fe9e1008484a4e5573b6b64c222d7e24b5e34e3b8bf79967595e041c67af3b3b9d6230c84eb14ec3556964e587260923485799a302573a8bbfd3e"
Audio-info: next URL: "https://stream-cdn-1.open.fm/OFM20/ngrp:standard/playlist.m3u8?t=cbb91a479b547ab4884e1ea7c11899d319925706a865e3a1ddc00e5aef96a73cacf30f047c70506b7c7b74df725dece7c15b976aeff1f2be8ad0a36f2782f8586c3786cb424328da1aff1d38808dcaacc6b1c295a26deda4d3f6901e97918663136727c67b815f1ff04ace3d0e87f7ff1f9fe669de2ebf072ac2cd34cb6966d86501d4321488c3767e7bc5a60cae5296fcbb186a55cb1fe9e1008484a4e5573b6b64c222d7e24b5e34e3b8bf79967595e041c67af3b3b9d6230c84eb14ec3556964e587260923485799a302573a8bbfd3e"
Audio-info: chunked data transfer
Audio-info: next URL: "https://stream-cdn-1.open.fm/OFM20/ngrp:standard/playlist.m3u8?t=cbb91a479b547ab4884e1ea7c11899d319925706a865e3a1ddc00e5aef96a73cacf30f047c70506b7c7b74df725dece7c15b976aeff1f2be8ad0a36f2782f8586c3786cb424328da1aff1d38808dcaacc6b1c295a26deda4d3f6901e97918663136727c67b815f1ff04ace3d0e87f7ff1f9fe669de2ebf072ac2cd34cb6966d86501d4321488c3767e7bc5a60cae5296fcbb186a55cb1fe9e1008484a4e5573b6b64c222d7e24b5e34e3b8bf79967595e041c67af3b3b9d6230c84eb14ec3556964e587260923485799a302573a8bbfd3echunklist_b192000.m3u8?t=cbb91a479b547ab4884e1ea7c11899d319925706a865e3a1ddc00e5aef96a73cacf30f047c70506b7c7b74df725dece7c15b976aeff1f2be8ad0a36f2782f8586c3786cb424328da1aff1d38808dcaacc6b1c295a26deda4d3f6901e97918663136727c67b815f1ff04ace3d0e87f7ff1f9fe669de2ebf072ac2cd34cb6966d86501d4321488c3767e7bc5a60cae5296fcbb186a55cb1fe9e1008484a4e5573b6b64c222d7e24b5e34e3b8bf79967595e041c67af3b3b9d6230c84eb14ec3556964e587260923485799a302573a8bbfd3e"
Audio-streamtitle: HTTP/1.1 403 Forbidden
Audio-info: Closing web file "https://getradio.reconv.pl/openfm?s=po-polsku-80"
Audio-info: connect to: "getradio.reconv.pl" on port 443 path "/openfm?s=po-polsku-80"
Audio-info: SSL has been established in 361 ms
Audio-lasthost: https://getradio.reconv.pl/openfm?s=po-polsku-80
Audio-info: Closing web file "https://getradio.reconv.pl/openfm?s=po-polsku-80"
```

In general, the issue is kinda random, because once a few tries, it succesfully connects and plays. I have no clue whats going on :(

schreibfaul1, 2025-09-24:
Hi Marcin,
please show me the first URL you visit.

Marcin-Fenger, 2025-09-24:
Hi, source link is: https://getradio.reconv.pl/openfm?s=po-polsku-80