schreibfaul1/ESP32-audioI2S #1171: After the recent updates, the sound from high-resolution FLAC stations is distorted

cniedzi, 2025-09-27:
Hi Wolle,

After the recent updates, the sound from high-resolution FLAC stations is distorted, i.e.:
[https://s2.audiostream.hu/roxy_FLAC](https://s2.audiostream.hu/roxy_FLAC)
[https://s2.audiostream.hu/bdpstrock_FLAC](https://s2.audiostream.hu/bdpstrock_FLAC)

Btw. Did you define SR_48K in Audio.h by default?

PS. Looks like the problems with the LovyanGFX dissapeared after the recent updates.

schreibfaul1, 2025-09-27:
I'm only testing with the S3. Have you set PSRAM to 120MHz?

cniedzi, 2025-09-27:
> I'm only testing with the S3. Have you set PSRAM to 120MHz?

Yep. Tommorow, I will check which commit introduced this issue at my end.
```
#
# ESP PSRAM
#
CONFIG_SPIRAM=y

#
# SPI RAM config
#
# CONFIG_SPIRAM_MODE_QUAD is not set
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
# CONFIG_SPIRAM_XIP_FROM_PSRAM is not set
# CONFIG_SPIRAM_FETCH_INSTRUCTIONS is not set
# CONFIG_SPIRAM_RODATA is not set
CONFIG_SPIRAM_SPEED_120M=y
# CONFIG_SPIRAM_SPEED_80M is not set
# CONFIG_SPIRAM_SPEED_40M is not set
# CONFIG_SPIRAM_TIMING_TUNING_POINT_VIA_TEMPERATURE_SENSOR is not set
CONFIG_SPIRAM_SPEED=120
# CONFIG_SPIRAM_ECC_ENABLE is not set
CONFIG_SPIRAM_BOOT_HW_INIT=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_PRE_CONFIGURE_MEMORY_PROTECTION=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
# CONFIG_SPIRAM_MEMTEST is not set
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
# CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP is not set
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY is not set
# CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY is not set
# end of SPI RAM config
# end of ESP PSRAM
```

cniedzi, 2025-09-27:
Strange, the simple & short program runs smoothly... I will test more tomorrow.

Btw. I think the outBuff in Audio_process_i2s for opus streams isn't provided anymore, take a look here:
```
Ogg Opus HTTP; http://streams.radiomast.io/ref-64k-ogg-opus-stereo
Ogg Opus HTTPS; https://streams.radiomast.io/ref-64k-ogg-opus-stereo
```

Marcin-Fenger, 2025-09-28:
In main release 3.4.3 flac crashes

audio_logs:   flac_decoder.cpp:931 Flac unimplemented reserved subtype: 21

audio_logs:   flac_decoder.cpp:931 Flac unimplemented reserved subtype: 16

audio_logs:   flac_decoder.cpp:797 Flac, reserved blocksize unsupported, block size code: 0

info: ....... syncword found at pos 2741

info: ....... syncword found at pos 0

info: ....... Channels: 2

info: ....... SampleRate (Hz): 44100

info: ....... BitsPerSample: 16

audio_logs:   flac_decoder.cpp:975 Flac reserved residual coding, method: 2

audio_logs:   flac_decoder.cpp:985 Flac, wrong rice partition number

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008de04  PS      : 0x00060e35  A0      : 0x80090af4  A1      : 0x3ffeab60  
  #0  0x4008de04 in esp_cpu_compare_and_set at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:200

cniedzi, 2025-09-28:
> I have just overwritten very recent files - flac_decoder and audiolib_Structs&psram_unique_ptr. It was memory allocation problem. Wolle added calloc_array function. Now it works well so far.

Unfortunately, it doesn't work for me :(

Marcin-Fenger, 2025-09-28:
Works flac but stream still cause crash at this moment

schreibfaul1, 2025-09-28:
Hi Marcin.
This appears to be a memory issue. Since the decoders are now class objects, some of the memory has moved from the stack to the heap. You can safely reduce the size of the Arduino loop stack to achieve more "free heap".
Place this in main.cpp at first `SET_LOOP_TASK_STACK_SIZE(8 * 1024);`
8192 DWORDs is the default. And if necessary, check with `SerialPrintfln("looptask ... stackHighWaterMark: %lu bytes", (long unsigned)uxTaskGetStackHighWaterMark(NULL) * 4);`

Hi cniedzi,
I saw your video. These appear to be brief dropouts. FLAC is the most demanding decoder with the lowest compression rate.
A task responsible for decoding runs within ausdioI2s. This is probably being slowed down. In platformio.ini, `configTASKLIST_INCLUDE_COREID=1` is set. You could try setting it to 0.

cniedzi, 2025-09-28:
Hi Wolle,

Unfortunately, setting `configTASKLIST_INCLUDE_COREID` to 0 hasn't helped :( It's strange, it was working perfectly fine before the commit "Better coefs".

Btw. Did you have a chance to look into why outBuff isn't provided for Opus streams?

schreibfaul1, 2025-09-28:
I set the coefs back to vevtor instead of a deque.

The outBuff is passed to each decoder; for Opus, it is here:

[image]

How satisfied are you with the P4? On my board (Waveshare ESP32P4 nano), the WiFi is provided by an ESP32C6 and the data throughput is extremely slow.

cniedzi, 2025-09-28:
With the latest commit these FLAC streams works good (I can hear very, very small crackles, but probably they were previously there).

As for the Opus I2S data, the problem was on me - I was looking for at least 1024 samles, while decoder returns only 960, thats why the spectrum analyzer wasn't drawn :)

closed 2025-09-28 by cniedzi